mbentley/docker-omada-controller #573: [Bug]: Port Nos 29815 and 29816 are not exposed in Dockerfile and Controller cannot connect to EAPs etc.

am4c130d, 2025-05-08:
### Controller Version

v5.15.20.20

### Describe the Bug

Firstly, the docker image is great - thank you.

I'm trying to use the Omada controller in standard docker bridge mode on a Docker Swarm cluster - V28.1.1ce

None of my devices (router, switch and EAPs) are discovered/connect, however switching to MACVLAN works perfectly. 

While troubleshooting I noticed that docker inspect doesn't show ports 29815 and 29816 in the exposed ports despite being in the compose/stack config, and and they aren't listed in the dockerfile either - which seems like the issue. 

This documentation from TP Link https://community.tp-link.com/en/business/kb/detail/362 indicates 29815 and 29816 are needed.

### Expected Behavior

I expect all my devices to be discovered and connected to the controller when the controller is running in a bridged container on docker swarm.

### Steps to Reproduce

1. Create docker stack file
2. Ensure all port number variables are listed as ports
3. Deploy stack
4. All existing devices will not connect to the controller

### How You're Launching the Container

```plain
version: "3.8"

services:
  omada-controller:
    image: mbentley/omada-controller:latest
    deploy:
      replicas: 1
    ulimits:
      nofile:
        soft: 4096
        hard: 8192
    stop_grace_period: 60s
    environment:
      - PUID=508
      - PGID=508
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTPS_PORT=8843
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - PORT_TRANSFER_V2=29815
      - PORT_RTTY=29816
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=Americas/New_York
    ports:
      - 8088:8088/tcp
      - 8043:8043/tcp
      - 8843:8843/tcp
      - 27001:27001/udp
      - 29810:29810/udp
      - 29811-29816:29811-29816/tcp
    volumes:
      - /var/data/ssd/omada/data:/opt/tplink/EAPController/data
[... 393 lines cut ...]
                    "IPPrefixLen": 24,
                    "IPv6Gateway": "",
                    "GlobalIPv6Address": "",
                    "GlobalIPv6PrefixLen": 0,
                    "DNSNames": [
                        "omada_omada-controller.1.8vuan9ulaug8km25gnxia4b6u",
                        "4d891f6e3f37",
                        "omada"
                    ]
                }
            }
        }
    }
]
```

### MongoDB Logs

```plain

```

### Additional Context

Before adding any ports, in MACVLAN mode, I was able to connect to the controller from TP Link's Cloud controller, but in the bridge mode this failed. Adding the ports enabled the controller in bridge mode to connect to the TP Link Cloud, but not the devices themselves. It's highly probable that ports that are not exposed are needed.

mbentley, 2025-05-08:
Hmm, this seems similar to https://github.com/mbentley/docker-omada-controller/discussions/562 where they're reporting issues with adoption.  They're also using swarm mode there as well.  Just curious - have you tried specifying the ports individually instead of the range just to make sure there isn't something funky with the port range?

mbentley, 2025-05-08:
I just added the ports in the Dockerfile in https://github.com/mbentley/docker-omada-controller/pull/574.  New images should be published shortly.  See if that makes any difference for you.

am4c130d, 2025-05-08:
Ahh, I was just about to confirm that breaking those ports out in to a list worked, but as I did an image pull to reload the container, it's using the new dockerfile  - the ports are now shown in docker inspect, which is good.

Unfortunately, all my devices still show as disconnected - it may be that they need to timeout from the old IP for the controller and then go through adoption etc. I'll give them an hour and then reboot the switch and gateway and see where we are. 

Thank you for the really fast response.

mbentley, 2025-05-08:
Yeah, I was trying to recall if swarm mode did anything unique with ports as it related to the Dockerfile vs at runtime - I didn’t think so but it didn’t hurt to add them.

am4c130d, 2025-05-08:
Update: after about an hour, the devices were still "disconnected". I rebooted the switch (all APs are POE from the switch), which didn't make any difference, they were still showing as disconnected. My conclusion is that Docker Swarm is not forwarding the broadcast packets, either at all, or to the right container - despite there being port information in the broadcast packet. 

Currently, I cannot use the Controller in bridge mode as the devices remain disconnected, but everything works fine in MACVLAN, but I have issues with MACVLAN which is my motivation to move to bridge/standard mode.

mbentley, 2025-05-09:
Someone pointed out [here](https://github.com/jkunczik/home-assistant-omada/issues/88#issuecomment-2864754580) that the controller is listing ports required and there was one missing from the documentation: 19810/udp.  Could you add that to the ports and see if it still fails?

am4c130d, 2025-05-09:
> Someone pointed out [here](https://github.com/jkunczik/home-assistant-omada/issues/88#issuecomment-2864754580) that the controller is listing ports required and there was one missing from the documentation: 19810/udp. Could you add that to the ports and see if it still fails?

Unfortunately that made no difference - my issue isn't that they cannot be adopted, all the nodes show as "Disconnected", which sounds like no connectivity at all.

> Could you try that and see if that helps the switch? I am not sure if the EAPs have the same capability of being able to manually set the inform URL.

I'll try that over the weekend, given it's a Docker Swarm cluster the IP address/URL will change, so not sure how robust that will be.

I think the issue is connectivity - which sounds like it's container/routing related. But I have other containers that are running bridge mode talking to devices on the network, and other people have your container working in bridge mode, so it may be my configuration. I'll keep probing there as well and report back. Please keep ideas coming.

Thank you again for your help.

mbentley, 2025-05-09:
Thanks for checking the port.  I can only conclude that the TP-Link folks must have broken something about the adoption process for certain devices - the one common thing among people experiencing issues is that they all seem to have switches.  The only thing I can figure is that it's like they broke something related to adoption and NAT traversal seeing as adoption works fine in host mode or macvlan mode but not bridged.  I just don't have the additional hardware to setup some sort of test environment to try to reproduce unfortunately.

am4c130d, 2025-05-14:
I was unable to test this weekend, but wonder if bridge mode will work especially on swarm. From the TP-Link docs, discovery is via broadcast packet, but bridge mode isn't really a bridge, so how would the broadcast packet be transferred over the NAT interface?

I'm sure there's a way to hack it, but I wonder if that is the problem?

mbentley, 2025-05-14:
Yeah, so I have been playing with this in a test lab environment and what I can see is that when you're in bridge mode now, it seems like you need to tell the EAP where the controller is through the Controller Inform URL.  I don't know when it changed or what they changed because it seems like tons of people had it working fine in bridge mode but it's not working any longer for several people.

I haven't tried to use the java tool mentioned here to set it: https://www.tp-link.com/us/support/faq/3087/  but I did test using the EAP's web interface to set it before adopting and it succeeded.  Unsure about in situations where the Swarm IP may change in terms of longevity of the devices staying connected.

am4c130d, 2025-05-14:
Good to know, thank you. 

I use a /32 subnet for Omada to "fix" the IP address. it's not really supported in swarm, as there can be only one subnet with a given gateway address per host - but as I have three hosts, and only two apps need MACVLAN, I can survive any one node failure/maintenance window with no service degradation - but it's a hack...

However, there's a bug/limitation in docker swarm [IPAM not coordinated in Swarm](https://github.com/moby/moby/issues/49921) that means if each docker swarm host has a MACVLAN range of a few address, and each host uses the same IP range, docker happily hands out conflicting IP addresses - i.e. each host hands out addresses with no regard to the other nodes. To work around that, I had a unique range per host, and as the Omada container was moved between hosts (by maintenance upgrades to the host), it's IP address changed - the switches, APs and gateways all seemed fine, they'd rediscover the controller and get adopted. But this bug was what got me looking at bridge mode. The real reason to fix the IP address is not for the devices, but for me, to find the web interface quickly.

am4c130d, 2025-05-14:
Ahh, just read the article and understand what you meant by changing IP address. It's a shame, as method 3 - DHCP option 138 would be great if bridge mode IP address could be locked down.

mbentley, 2025-05-15:
Oh shoot, I'd forgotten that you can't specify an IP address with Swarm mode on a service.  That's a pain and really unfortunate that Swarm development has all but stopped, as far as I am aware.

I updated the README and known issues in https://github.com/mbentley/docker-omada-controller/pull/579 with details with that FAQ about needing to set the controller inform URL.

mbentley, 2025-05-15:
I am going to go ahead and close this but let me know if there are any persistent issues.

closed 2025-05-15 by mbentley