mbentley/docker-omada-controller #598: [Bug]: No connectivity to any Devices on Mac Docker

driveformer, 2025-07-21:
### Controller Version

V5.15.24.18

### Describe the Bug

I’ve been using Omada for on set WiFi and mesh, and I power cycle frequently. However about a week ago my instance stopped detecting any of my devices. Docker reported it needed to repair something (I don’t remember what now) but I’m wondering if this is a docker issue, Mac issue, etc if anyone has insight? I tried updating from what I was running to the latest and no change. Only thing that did oddly change anything was setting the IP for the controller incorrectly to another device brought back one AP, letting it set to the container or to the Mac’s IP shows nothing. Thanks!

### Expected Behavior

Reboot should show all of the already paired and functioning devices.

### Steps to Reproduce

Restarting container or Mac leads to same issue.

### How You're Launching the Container

```plain
Just the default play button. Also, ports have been set.
```

### Container Logs

```plain
N/A atm
```

### MongoDB Logs

```plain
N/A atm
```

### Additional Context

I apologize, I don’t have access to pull logs as I don’t want to interfere with the other work actively on the machine.

mbentley, 2025-07-21:
Without logs or how you’re launching the container, I can only guess as to what you’re seeing.  Seems that recently there were changes that made it so the devices required to be told what the IP or host name is of the controller. I’m assuming your mac device running the container hasn’t changed IPs?

driveformer, 2025-07-28:
> I've had what seems to be the same problem since about May this year. I
> don't have records of everything that's happened and it's taken me some
> time to get round to trying to sort out the problem which doesn't affect
> the correct operation of my Omada network.
>
> My network consists of an ER605 gateway, an SG2210P PoE switch, 4
> EAP615-Wall APs, and an EAP650 ceiling mount AP. The gateway connects to my
> fibre connection.
>
> I'm running a containerised Omada controller (5.15.20.18) under Docker
> Desktop (v43.43.2) on an M1 Mac Mini running Sequoia 15.5.
>
> Until 6th May my set-up was working perfectly. Since then all my Omada
> devices are showing up as DISCONNECTED.
> All the devices show up when I scan my LAN, they all have reasonable IP-V4
> IP addresses. I can log in to the devices through their web interface. Some
> devices (e.g. ER605) tell me that it "This Gateway has been managed by
> Controller before. You can reset your device to manage it as a standalone
> Gateway.", others offer me the option of rebooting or resetting. I have
> rebooted some devices - they do not connect to the controller.
>
> I have powered cycled some devices and at least some of them then
> connected to the controller - but only for a short time.
>
> I've now got some time available to try to sort out this problem. I've no
> knowledge of Docket beyond what I picked up when following the instructions
> on how to use the Omada controller container, and what I've read when
> hunting possible solutions.
>
> Any advice on how to proceed would be greatly appreciated.

Thank you so much Roger! I’m actively using it in a working film
environment so I have little down time to actually investigate further but
my setup is basically identical to yours.

Thanks!

driveformer, 2025-07-28:
> I've been trying to find the latest version of Docker Desktop that works
> and it seems to be 4.40.0 (31/3/25).
>
> 4.38.0 works, 4.39.0 has problem with my browser not being able to connect
> to the the controller, 4.40.0 works fine, 4.41.0 shows the "disconnected"
> problem. I can't see anything obvious in the change log what might have
> caused this and I don't think I've enough skill to debug this any further.
> I'll stick with 4.40.0 for the moment.

Thank you! 4.40 is also working for me again. Saved my butt

Roger-Shepherd, 2025-07-29:
> I was taking a look at the [release notes](https://docs.docker.com/desktop/release-notes/#4410) for Docker Desktop and I don't see anything that completely sticks out besides maybe "Fixed a bug that was causing outgoing UDP connections to be eagerly closed." and I don't see anything really sticking out for the [Docker Engine release notes](https://docs.docker.com/engine/release-notes/28/#2811) of the version that it was bumped to either.

I thought that might be why 4.40 works but 4.39 didn't. 

I'm afraid I think I'm at the end of what I can do - although I guess I could run up a recent version and post the log if anyone would find that useful

Roger-Shepherd, 2025-11-04:
At the moment I'm sticking to Sequoia but I would like to to upgrade the OS now that 26.1 is out. Is there any information about compatibility with 26? I know that Docker had some performance issues but they've been fixed in a recent release of desktop - however, I'm still on 4.40 (because later versions give me this connectivity problem).

closed 2025-11-09 by mbentley