mbentley/docker-omada-controller #599: [Bug]: omada-controller will not start

bensonmunyan, 2025-07-22:
### Controller Version

v5.15.24.18

### Describe the Bug

Docker appears to enter a startup loop and will not initiate. Currently running on Unraid. I did check the current issues, but I am (apparently) illiterate and cannot determine if this is an existing issue, though it sounds similar.

### Expected Behavior

Docker runs successfully. 

### Steps to Reproduce

Start the container.

### How You're Launching the Container

```plain
Start docker container manually via GUI or auto-start. Both fail.
```

### Container Logs

```plain
07-22-2025 16:30:32.692 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
07-22-2025 16:30:32.697 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
07-22-2025 16:30:32.732 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
07-22-2025 16:30:32.740 INFO [main] [] c.t.s.o.c.o.a(): Native Image: null
07-22-2025 16:30:32.744 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): spring.index.ignore : true
07-22-2025 16:30:32.744 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
07-22-2025 16:30:32.748 INFO [main] [] c.t.s.o.s.g.c(): Current dir omadac type is linux
07-22-2025 16:30:32.762 INFO [main] [] c.t.s.o.c.o.c.c(): success to load configuration omada.properties
07-22-2025 16:30:32.763 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
07-22-2025 16:30:32.763 INFO [main] [] c.t.s.o.s.g.c(): Configuration omadacType is linux
07-22-2025 16:30:32.772 INFO [main] [] c.t.s.o.s.g.f(): Start startup message daemon.
07-22-2025 16:30:32.802 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
07-22-2025 16:30:32.805 INFO [main] [] c.t.s.o.m.c.c.d(): FailoverClusterEnable is false
07-22-2025 16:30:33.309 ERROR [main] [] c.t.s.o.s.s.b(): Command execution failed with exit code 100 .
07-22-2025 16:30:33.310 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Fail to start mongo DB server with exit code 100
07-22-2025 16:30:33.310 WARN [main] [] c.t.s.o.s.s.b(): Fail to start mongo DB server
07-22-2025 16:30:33.310 WARN [main] [] c.t.s.o.s.s.a(): com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
	at com.tplink.smb.omada.starter.server.b.a(SourceFile:210) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.server.a.e(SourceFile:74) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:448) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaBootstrap.w(SourceFile:641) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaBootstrap.a(SourceFile:212) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaBootstrap.f(SourceFile:494) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:121) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:68) ~[local-starter-5.15.24.18.jar:5.15.24.18]
07-22-2025 16:30:36.246 ERROR [main] [] c.t.s.o.s.s.b(): Command execution failed with exit code 100 .
07-22-2025 16:30:36.246 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Fail to start mongo DB server with exit code 100
07-22-2025 16:30:36.246 WARN [main] [] c.t.s.o.s.s.b(): Fail to start mongo DB server
07-22-2025 16:30:36.246 WARN [main] [] c.t.s.o.s.s.a(): com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
	at com.tplink.smb.omada.starter.server.b.a(SourceFile:210) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.server.a.e(SourceFile:74) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:448) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaBootstrap.t(SourceFile:441) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaBootstrap.w(SourceFile:644) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaBootstrap.a(SourceFile:212) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaBootstrap.f(SourceFile:494) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:121) ~[local-starter-5.15.24.18.jar:5.15.24.18]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:68) ~[local-starter-5.15.24.18.jar:5.15.24.18]
07-22-2025 16:30:50.248 INFO [mongo-check-thread] [] c.t.s.o.s.OmadaBootstrap(): mongod was shutdown, restarting it
07-22-2025 16:30:50.252 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: An unexpected error occurred in MongoDb. Omada Controller will try to restart it.
07-22-2025 16:30:51.180 ERROR [mongo-check-thread] [] c.t.s.o.s.s.b(): Command execution failed with exit code 100 .
```

[... 154 lines cut ...]
2024-09-23T13:27:54.165-0400 I NETWORK  [initandlisten] removing socket file: /tmp/mongodb-27217.sock
2024-09-23T13:27:54.165-0400 I CONTROL  [initandlisten] now exiting
2024-09-23T13:27:54.165-0400 I CONTROL  [initandlisten] shutting down with code:100
2024-09-23T13:29:35.833-0400 I CONTROL  [main] ***** SERVER RESTARTED *****
2024-09-23T13:29:35.836-0400 I CONTROL  [initandlisten] MongoDB starting : pid=218 port=27017 dbpath=../data/db 64-bit host=UNRAID
2024-09-23T13:29:35.836-0400 I CONTROL  [initandlisten] db version v3.6.8
2024-09-23T13:29:35.836-0400 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
2024-09-23T13:29:35.836-0400 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
2024-09-23T13:29:35.836-0400 I CONTROL  [initandlisten] allocator: tcmalloc
2024-09-23T13:29:35.837-0400 I CONTROL  [initandlisten] modules: none
2024-09-23T13:29:35.837-0400 I CONTROL  [initandlisten] build environment:
2024-09-23T13:29:35.837-0400 I CONTROL  [initandlisten]     distarch: x86_64
2024-09-23T13:29:35.837-0400 I CONTROL  [initandlisten]     target_arch: x86_64
2024-09-23T13:29:35.837-0400 I CONTROL  [initandlisten] options: { repair: true, storage: { dbPath: "../data/db" }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2024-09-23T13:29:35.838-0400 I STORAGE  [initandlisten] Detected WT journal files.  Running recovery from last checkpoint.
2024-09-23T13:29:35.838-0400 I STORAGE  [initandlisten] journal to nojournal transition config: create,cache_size=15468M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),cache_cursors=false,compatibility=(release="3.0",require_max="3.0"),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
2024-09-23T13:29:36.210-0400 E STORAGE  [initandlisten] WiredTiger error (2) [1727112576:210182][218:0x14b123a9fac0], file:WiredTiger.wt, connection: ../data/db/WiredTiger.wt: handle-open: open: No such file or directory
2024-09-23T13:29:36.210-0400 E -        [initandlisten] Assertion: 28718:2: No such file or directory src/mongo/db/storage/wiredtiger/wiredtiger_kv_engine.cpp 397
2024-09-23T13:29:36.211-0400 I STORAGE  [initandlisten] exception in initAndListen: Location28718: 2: No such file or directory, terminating
2024-09-23T13:29:36.211-0400 I NETWORK  [initandlisten] shutdown: going to close listening sockets...
2024-09-23T13:29:36.211-0400 I NETWORK  [initandlisten] removing socket file: /tmp/mongodb-27017.sock
2024-09-23T13:29:36.211-0400 I CONTROL  [initandlisten] now exiting
2024-09-23T13:29:36.211-0400 I CONTROL  [initandlisten] shutting down with code:100
2024-09-23T13:29:36.224-0400 I CONTROL  [main] ***** SERVER RESTARTED *****
```

Stackhollow, 2025-07-28:
Hey,
I got exactly the same issue on my container startup. Thought it must have been switching hardware from SSD to NVMe. But even fresh install without any perstistant volumes won't work. I also tried other releases, backups up to february or pruned the controller image completeley, but still the same issue.

Got no clue what this is about.

```
07-28-2025 17:56:09.645 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: start the omada controller
07-28-2025 17:56:09.647 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: set property finished
07-28-2025 17:56:09.662 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): record: configure log finished
07-28-2025 17:56:09.667 INFO [main] [] c.t.s.o.c.o.a(): Native Image: null
07-28-2025 17:56:09.669 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap prepare
07-28-2025 17:56:09.672 INFO [main] [] c.t.s.o.s.g.c(): Current dir omadac type is linux
07-28-2025 17:56:09.683 INFO [main] [] c.t.s.o.c.o.c.c(): success to load configuration omada.properties
07-28-2025 17:56:09.683 INFO [main] [] c.t.s.o.c.o.OmadacType(): omadacType: Local Controller
07-28-2025 17:56:09.683 INFO [main] [] c.t.s.o.s.g.c(): Configuration omadacType is linux
07-28-2025 17:56:09.688 INFO [main] [] c.t.s.o.s.g.f(): Start startup message daemon.
07-28-2025 17:56:09.727 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
07-28-2025 17:56:09.729 INFO [main] [] c.t.s.o.m.c.c.d(): FailoverClusterEnable is false
2025-07-28T17:56:09.742+0200 I CONTROL  [main] ***** SERVER RESTARTED *****
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten] MongoDB starting : pid=247 port=27217 dbpath=../data/db 64-bit host=server
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten] db version v3.6.8
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten] allocator: tcmalloc
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten] modules: none
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten] build environment:
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten]     distarch: x86_64
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten]     target_arch: x86_64
2025-07-28T17:56:09.746+0200 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db" }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2025-07-28T17:56:09.747+0200 I -        [initandlisten] Detected data files in ../data/db created by the 'wiredTiger' storage engine, so setting the active storage engine to 'wiredTiger'.
2025-07-28T17:56:09.747+0200 I STORAGE  [initandlisten] 
2025-07-28T17:56:09.747+0200 I STORAGE  [initandlisten] ** WARNING: Using the XFS filesystem is strongly recommended with the WiredTiger storage engine
2025-07-28T17:56:09.747+0200 I STORAGE  [initandlisten] **          See http://dochub.mongodb.org/core/prodnotes-filesystem
2025-07-28T17:56:09.747+0200 I STORAGE  [initandlisten] wiredtiger_open config: create,cache_size=15425M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),cache_cursors=false,compatibility=(release="3.0",require_max="3.0"),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
2025-07-28T17:56:09.895+0200 I STORAGE  [initandlisten] WiredTiger message [1753718169:895210][247:0x7b417b74cac0], txn-recover: Main recovery loop: starting at 4/6400
2025-07-28T17:56:09.943+0200 I STORAGE  [initandlisten] WiredTiger message [1753718169:943636][247:0x7b417b74cac0], txn-recover: Recovering log 4 through 5
2025-07-28T17:56:09.969+0200 I STORAGE  [initandlisten] WiredTiger message [1753718169:969532][247:0x7b417b74cac0], txn-recover: Recovering log 5 through 5
2025-07-28T17:56:09.993+0200 I STORAGE  [initandlisten] WiredTiger message [1753718169:993945][247:0x7b417b74cac0], txn-recover: Set global recovery timestamp: 0
2025-07-28T17:56:10.001+0200 I CONTROL  [initandlisten] 
2025-07-28T17:56:10.001+0200 I CONTROL  [initandlisten] ** WARNING: Access control is not enabled for the database.
2025-07-28T17:56:10.001+0200 I CONTROL  [initandlisten] **          Read and write access to data and configuration is unrestricted.
2025-07-28T17:56:10.001+0200 I CONTROL  [initandlisten] 
2025-07-28T17:56:10.006+0200 I FTDC     [initandlisten] Initializing full-time diagnostic data capture with directory '../data/db/diagnostic.data'
2025-07-28T17:56:10.006+0200 I NETWORK  [initandlisten] waiting for connections on port 27217
07-28-2025 17:56:10.234 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 247
07-28-2025 17:56:10.237 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
07-28-2025 17:56:10.237 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started
07-28-2025 17:56:40.376 WARN [main] [] c.t.s.o.s.OmadaBootstrap(): Timed out after 30000 ms while waiting to connect. Client iew of cluster state is {type=UNKNOWN, servers=[{address=127.0.0.1:27217, type=UNKNOWN, state=CONNECTING, exception={com.mongodb.MongoSocketOpenException: Exception opening socket}, caused by {java.net.SocketTimeoutException: Connect timed out}}]
[... cut ...]
	at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:62) ~[local-starter-5.15.20.20.jar:5.15.20.20]
07-28-2025 17:56:40.383 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap record finished
07-28-2025 17:56:40.383 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: bootstrap startup
07-28-2025 17:56:40.403 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: start run omada tasks
07-28-2025 17:56:40.403 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(): record: task SpringBootStartupTask start
```

kingp0dd, 2025-08-08:
can't figure it out. ihave the WiredTiger.wt  but still get 
```

omada-controller  | 08-08-2025 19:39:16.724 INFO [mongo-check-thread] [] c.t.s.o.s.OmadaBootstrap(): mongod was shutdown, restarting it
omada-controller  | 08-08-2025 19:39:16.724 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: An unexpected error occurred in MongoDb. Omada Controller will try to restart it.
omada-controller  | 08-08-2025 19:39:17.327 ERROR [mongo-check-thread] [] c.t.s.o.s.s.b(): Command execution failed with exit code 100 .
omada-controller  | 08-08-2025 19:39:17.327 ERROR [mongo-check-thread] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Fail to start mongo DB server with exit code 100
omada-controller  | 08-08-2025 19:39:17.327 WARN [mongo-check-thread] [] c.t.s.o.s.s.b(): Fail to start mongo DB server
omada-controller  | 08-08-2025 19:39:17.327 WARN [mongo-check-thread] [] c.t.s.o.s.s.a(): com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
omada-controller  | com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
omada-controller  | 	at com.tplink.smb.omada.starter.server.b.a(SourceFile:210) ~[local-starter-5.15.24.19.jar:5.15.24.19]
omada-controller  | 	at com.tplink.smb.omada.starter.server.a.e(SourceFile:74) ~[local-starter-5.15.24.19.jar:5.15.24.19]
omada-controller  | 	at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:448) ~[local-starter-5.15.24.19.jar:5.15.24.19]
omada-controller  | 	at com.tplink.smb.omada.starter.OmadaBootstrap.t(SourceFile:441) ~[local-starter-5.15.24.19.jar:5.15.24.19]
omada-controller  | 	at com.tplink.smb.omada.starter.OmadaBootstrap.b(SourceFile:722) ~[local-starter-5.15.24.19.jar:5.15.24.19]
omada-controller  | 	at java.base/java.lang.Thread.run(Thread.java:840) [?:?]
```

awptechnologies, 2025-08-09:
I ended up saying screw it. I still run in docker swarm but i just reset controller and set everything up in host mode and didn't mess with any bind mounts and after re-configuring everything it works perfect. I just deleted all the associated volumes and started fresh.

mbentley, 2025-11-09:
I'm going to go ahead and close this for now - happy to re-open if needed.